Open-Smartwatch/3d-files #13: The display flex ribbon extends more than the case allows for

MarkJeronimus, 2021-05-17:
The Light Edition watch I have doesn't fit (without squeezing) into the case.

I measure that the case allows for a 38.8mm total dimension from the very top of the display (i.e. 18.0mm above the optical center) to the bottom-most extend of the cavity for the flex. When I measure my item, the round display is exactly 36.0mm wide (just as a verifying measure to show that I don't measure wrong) and the total height is 39.5mm. This is 0.7mm too much. Given that the bend radius of the flex is about 0.75mm, this is no small issue. It _could_ probably fit, but it would seriously hurt the flex by squeezing it into a square shape with bend radii way too small.

This is a rendering of the STL file, along with a model of the electronics according to my measurements:

[image]

Note that the cavity for the flex also seems too shallow, as my flex measures about 1.95mm in thickness (in the model I used 2mm, at line 20)

uvwxy, 2021-05-20:
Best approach to fix this would be with the parameterised model: https://github.com/Open-Smartwatch/3d-files/tree/master/case-light-round-parametric-printed-straps

ShandilyaSolutions, 2021-09-18:
what if the case itself is designed by subtracting the dimensions of the pcb?  If provided with all the necessary data and dimensions, I can provide an elegant solution for this problem without falling in the need of all parameterized modelling. (Finger Crossed)